khoih-prog/AsyncWebServer_RP2040W #2: LED state appears to be reversed.

revell1, 2022-09-11:
### Describe the bug

In various examples, the use of the LED appears to be inverted.
i.e.
define LED_OFF             HIGH
define LED_ON              LOW

It looks from the code as if the LED should be OFF while IDLE and turn on while processing a request.
Where in fact the observed state of the LED looks like it is on ALL the time, the blink to OFF is so short that it is masked by the brightness of the LED.

### Steps to Reproduce

Build and run any of the examples where LED is used, and observe LED state.

### Expected behavior
As described above, would expect LED to be reversed.

### Actual behavior

As described above, the LED appears to be ON all the time.

### Debug and AT-command log (if applicable)

Used firefox to access target web address to cause request activity.

### Screenshots

N/A

### Information

* Arduino IDE version 1.8.19
* `RP2040` Core Version RP2040 core v2.5.2
* `RP2040` Board type RASPBERRY_PI_PICO_W
* Contextual information - Just trying out the library to see if it fits with my needs.
* Simplest possible steps to reproduce - build and run Async_AdvancedWebServer example program.
* Anything that might be relevant in your opinion, such as:
  * Operating system code built on Windows 10 laptop
  * Network configuration - Firefox 104.0.2 (64-bit) browsing to Pico-W IP address.

khoih-prog, 2022-09-11:
HI @revell1 

Thanks for spotting the bug. It'll be fixed in next release.

Regards,

khoih-prog, 2022-09-11:
Hi @revell1 

The new [**AsyncWebServer_RP2040W releases v1.0.2**](https://github.com/khoih-prog/AsyncWebServer_RP2040W/releases/tag/v1.0.2) has just been published. Your contribution is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/AsyncWebServer_RP2040W#contributions-and-thanks)

Best Regards,

---

### Release v1.0.2

1. Fix LED bug in examples

closed 2022-09-11 by khoih-prog