khoih-prog/AsyncWebServer_RP2040W #5: Can't connect to AsyncWebSocketServer_RP2040 via javascript

drmue, 2022-10-09:
Can't connect to AsyncWebSocketServer via javascript, tried with Firefox 105.0 in Ubuntu 20.02 LTS and Firefox 98.0.2 on Debian 10. 
But I'm able to connect via Python 3.7.9. 

1. Upload "PicoWAsyncSockAbare.ino" to RP pico W.

2. Open Serial Monitor to get IP-adress.

3. Try to connect with "AsyncSocCl.py".
    (This will work fine)

4. Try to connect with Webbrowser (Firefox)..
    (It does not work)

Am I do something wrong?

I think, I did understand AsyncWebServer_RP2040 is based on ESPAsyncWebServer.
After uploading ESP32AsyncSockAbare (same code) to ESP32 - Board (NodeEsp), connection is established with Webbrowser and with Python.

[AsyncWebServer_RP2040_Issue.zip](https://github.com/khoih-prog/AsyncWebServer_RP2040W/files/9741767/AsyncWebServer_RP2040_Issue.zip)

Arduino IDE version: 1.8.19
RP2040 core v2.6.0
RASPBERRY_PI_PICO_W Module
OS: Ubuntu 20.04 LTS
Linux xy-Inspiron-3593 5.15.0-48-generic #54~20.04.1-Ubuntu SMP Thu Sep 1 16:17:26 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

khoih-prog, 2022-10-10:
Hi @drmue 

Thanks for the bug report. This **`elusive very-hard-to-find bug`** has been identified and will be fixed in the next version, to be released by tomorrow.

Best Regards,

drmue, 2022-10-10:
Dear Khoih, 
thank you for your quick reply, which makes me very optimistic that I can continue my current projects very soon.
Best regards
drmue

khoih-prog, 2022-10-10:
Hi @drmue 

The new [**AsyncWebServer_RP2040W releases v1.3.0**](https://github.com/khoih-prog/AsyncWebServer_RP2040W/releases/tag/v1.3.0) has just been published. Your contribution is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/AsyncWebServer_RP2040W#contributions-and-thanks)

Good Luck with your exiting projects. We'd appreciate it if you can share some **public** info when projects done successfully.

Best Regards,

---

#### Release v1.3.0

1. Fix `crash` when using `AsyncWebSockets server`. Check [Can't connect to AsyncWebSocketServer_RP2040 via javascript #5](https://github.com/khoih-prog/AsyncWebServer_RP2040W/issues/5)
2. Add example [Async_WebSocketsServer](https://github.com/khoih-prog/AsyncWebServer_RP2040W/tree/main/examples/Async_WebSocketsServer) to demo the AsyncWebSockets Server with a `Python` [WSClient.py](https://github.com/khoih-prog/AsyncWebServer_RP2040W/tree/main/examples/Async_WebSocketsServer/WSClient_Python/WSClient.py)

---

[image]

closed 2022-10-10 by khoih-prog