khoih-prog/AsyncWebServer_RP2040W #12: WS: catchAll handler not working

roma2580, 2023-01-29:
Many thanks for this great library ! I was able to port an existing not-too-small ESP8266 project to RP within few hours. This project even includes Web sockets, and everything seemed to work fine until I happened to enter a non-existing url. 
In order to exclude any self-made nonsense, I tried the same situation with the example program that comes with the library, and found the same problem.  

### Describe the bug

There is no visible response in the browser when requesting a non-existing url. 

### Steps to Reproduce

Compile the original example Async_AdvancedWebServer.ino, nothing changed but ssid[] and pass[].
Enter the IP in the browser (e.g. 192.168.100.121) and see the correct page, with the nice animated diagram.
Enter the prepared alternative url (e.g. 192.168.100.121/inline) and see the correct response text "This works as well".
But:
Try any other url (e.g. 192.168.100.121/xyz.htm) and see nothing but a blank screen! The Firefox debugging tool explains that the http response code is 501, and content length is 0.
 
### Expected behavior

Looking into the function handleNotFound() in the example code, we would expect a text like "File Not Found", the uri and the http method GET. Moreover, the http response should be 404.

### Actual behavior

(see "steps to reproduce")

### Debug and AT-command log (if applicable)

(n.a.)

### Screenshots

(n.a.)

### Supporting information:

Library v1.4.1
Arduino IDE 1.8.19
RP2040 core v2.7.1
RASPBERRY_PI_PICO_W
Windows 7 64bit
Dell Latitude E6530

khoih-prog, 2023-01-30:
HI @roma2580 

Thanks for your nice encouraging words, and glad the library can somehow help you. Actually most of the credits must go to the original author(s) of ESPAsyncWebServer, I've just modified a little bit to port to RP2040W.

I actually can duplicate the bug and will spend some time to locate and fix, if possible and time permitted. Luckily the bug is not fatal and prevent the operation of the AsyncWebServer.

Will try fixing soon. If you can help somehow, I'd appreciate it.

Regards,

roma2580, 2023-01-30:
Hi Khoi Hoang,

thank you for your immediate response !
If that is of any help, I can confirm that the default catchAll handler
does work with the original library ESPAsyncWebSrv (v1.2.3) !
I just tried the library example simple_server.ino on an ESP8266 module..
Regards, Robert

roma2580, 2023-01-30:
Hi Khoi Hoang,

I guess I found the problem myself:
In File AsyncWebServer_RP2040W.cpp, look at the bottom of function
void AsyncWebServer::_attachHandler(AsyncWebServerRequest *request)
which is defined around line 182. While in your version there is
"request->setHandler(NULL);", in the original ESP version there is
"request->setHandler(_catchAllHandler);" !
I gave it a try, and - voilá, the catchAllHandler works ! Seems to be an
intermediate test version, and you just forgot to finally change it back.
Of course, I have no idea of any side effects.

Kind regards, Robert

khoih-prog, 2023-01-30:
Hi @roma2580 

The [**AsyncWebServer_RP2040W** v1.5.0](https://github.com/khoih-prog/AsyncWebServer_RP2040W/releases/tag/v1.5.0) has just been published. Your contribution is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/AsyncWebServer_RP2040W#contributions-and-thanks).

Please continue your great work of finding and fixing bugs. **Waiting for many more of your bug-report / PRs**. 

Best Regards,

---

#### Release v1.5.0

1. Fix `_catchAllHandler` not working bug. Check [catchAll handler not working #12](https://github.com/khoih-prog/AsyncWebServer_RP2040W/issues/12)

closed 2023-01-30 by khoih-prog